WildernessLabs/Netduino.Foundation #17: SerialLCD Baud rate should be configurable

bryancostanich, 2018-01-27:
according to the [datasheet](http://www.sparkfun.com/datasheets/LCD/SerLCD_V2_5.PDF) to set the baud, you send the control character and then a byte:

 * 2400 baud: <control>k
 * 4800 baud: <control>l
 * 9600 baud: <control>m
 * 14400 baud: <control>n
 * 19200 baud: <control>o
 * 38400 baud: <control>p

where would we do this? would it be in the constructor before we set the size? how would that work, would you open a serial connection at the speed you wanted, e.g. 38400, and then configure the lcd screen? or would you have to open a port at 9600, then configure, and close and reopen?

bryancostanich, 2018-01-27:
It sounds like (it's a bit ambiguous), from the [tutorial page](https://www.sparkfun.com/tutorials/246) that you might need to change the baud and then reconnect.

So we should discuss this. Maybe it's not worth it. While i can see the usefulness of this (i'm running into speed issues when updating quickly), i can also see people putting their LCD screens into a non default baud rate and then not being able to communicate with them.

NevynUK, 2018-01-28:
I purposely left this out of the implementation as this is normally a one time operation and can be achieved using an FTDI connector if really needed.

For most purposes, 9,600 baud should be adequate and this is the default rate when the device is shipped.  It also offers a reasonable balance between speed and reliability.

closed 2018-02-05 by bryancostanich