WildernessLabs/Netduino.Foundation #53: I2C Communication between Netduino and Arduino

vberkaltun, 2018-07-10:
Hi there,

Is it possible to communicate a Netduino 3 WiFi with an Arduino UNO on **`I2C`** bus? I wrote an example code (i2c scanner) based on **`Netduino.Foundation.Core.Samples`** for that and tried it but not work!

The all lines have an **`4.7 KΩ`**  pull up resistor.
Also GND pins are connected common with together.

**`Netduino side as master:`**
```C#
public static void Main() {
 // setup some configs
 ushort speed = 100;
 ushort timeout = 100;
 ushort numberOfDevices = 0;

 // loop forever
 while (true) {
  numberOfDevices = 0;

  // loop through all the possible I2C addresses (0-127)
  for (byte i = 0; i < 127; i++) {
   I2CDevice.Configuration configuration = new I2CDevice.Configuration(i, speed);
   I2CDevice device = new I2CDevice(configuration);

   try {
    byte[] data = {
     0
    };
    device.Config = configuration;
    I2CDevice.I2CTransaction[] transaction = {
     I2CDevice.CreateWriteTransaction(data)
    };

    var retryCount = 0;
    while (device.Execute(transaction, timeout) != data.Length) {
     if (retryCount > 3) {
      throw new Exception("WriteBytes: Retry count exceeded.");
     }
     retryCount++;
    }

    Debug.Print("Found I2C device at: " + i.ToString("X"));
    numberOfDevices++;
   } catch (Exception) {}
  }

  if (numberOfDevices == 0) {
   Debug.Print("No I2C devices found.");
  }

  // wait five seconds before scanning again.
  Thread.Sleep(5000);
 }
}
```

**`Arduino side as slave:`**
```C++
#include <Wire.h>

void setup() {
  Wire.setClock(100000);
  Wire.begin(8);
  Wire.onReceive(receiveEvent);
  
  Serial.begin(9600);
  Serial.println("Started");
}

void loop() {
}

void receiveEvent(int howMany) {
  while (1 < Wire.available()) {
    char c = Wire.read();
    Serial.print(c);
  }
  int x = Wire.read();
  Serial.println(x);
}
```

- Berk

IOT-123, 2018-07-11:
Hi vberkaltun,

I have a Netduino 3 WiFi and some ATTINY85 I2C slaves I was going to test.

In your code, I'm getting:
The type or namespace name 'I2CDevice' could not be found (are you missing a using directive or an assembly reference?)
on line:
`I2CDevice device = new I2CDevice(configuration);`

I expect you are just not finding slaves:
>  for that and tried it but not work!

My references are:
```
using System;
using System.Threading;
using Microsoft.SPOT;
using Netduino.Foundation.Communications; // grayed out - not in use
```

I used the nuget package today for the first time.

More details that may pinpoint my problem (I want to verify yours):

`The call is ambiguous between the following methods or properties: 'Netduino.Foundation.NumericExtensions.Clamp(double, double, double)' and 'Netduino.Foundation.NumericExtensions.Clamp(double, double, double)'	Netduino.Foundation	D:\_GitHub\Netduino.Foundation\Source\Netduino.Foundation\Color.cs	79`

Cheers,
Nic

NevynUK, 2018-07-11:
vberkaltun,

I have modified the code slightly by changing:

```csharp
catch (Exception) {}
```

to
```csharp
catch (Exception) {}
device.Dispose();
```

In doing this I get a number of exception error messages (as you would expect when a device does not exist).  I do get a message `Found I2C device at: 27` when I have the MCP23008 connected to the N3W board.

For information, the `I2CBus` class has been written to allow multiple configurations but the `I2CDevice` class does not.

Regards,
Mark

closed 2018-07-11 by NevynUK

vberkaltun, 2018-07-11:
Hi again,

I tried all of solutions but the result is still same - not work. Thank you for all.
I have also tested this master code with different slaves - BMP180 and BMP085. Not work again.

Now I am thinking about **`MultiI2C Lib`**, **`.NET Micro Framework Toolbox`** and **`I2CBus`** Lib. I'll try all of this libraries and after When Everything goes well I'll provide feedback for here.

Regards,
Berk

NevynUK, 2018-07-11:
vberkaltun,

Please post a link to the boards you are trying to use.

Regards,
Mark

vberkaltun, 2018-07-11:
The boards that I used is given at below:
 
1. [Arduino Uno R3 Compatible ATmega328P Microcontroller Card](https://www.amazon.com/RoboGets-Compatible-ATmega328P-Microcontroller-Electronics/dp/B01N4LP86I/ref=sr_1_7?ie=UTF8&qid=1531298361&sr=8-7&keywords=arduino+uno)
2. [BMP180 Digital Barometric Pressure Sensor Module](https://www.amazon.com/Barometric-Pressure-Compatible-Atomic-Market/dp/B00TMJP4JC)
3. [BMP085 Atmospheric Pressure Sensor Module](https://www.banggood.com/GY-68-300-1100hPa-BMP180-BOSCH-Temperature-Module-Atmospheric-Pressure-Sensor-Instead-Of-BMP085-p-1175978.html?rmmds=buy&cur_warehouse=CN)

In additional, I2C connection diagram is same as [the **`I2C/IIC`** article on **`wildernesslabs.co`**](http://developer.wildernesslabs.co/Netduino/Input_Output/Digital/I2C/).

[image]

Both of SCL - SDA have an **`4.7 KΩ`**  pull up resistor.
GND pins are connected with one another.

Regards,
Mark

NevynUK, 2018-07-11:
Looking at the BMP180 board (link 2), it already has 4K7 pull-ups on the board and so external pull-ups are not required.

Looking at the other board, it appears to have a very similar layout.  I suspect that it will also have pull-up resistors on the board as well.

From the markings on the SOT23 device on board (link 2) I would hazard a guess that the device is a 3.3V regulator (hence the input voltage range).  It looks like it could be a LM6206N3 configured to output 3.3V.

**Without a full circuit diagram and the list of part numbers then the following is what I can glean and you would need to proceed at your own risk.**

So, concentrating on the BMP180 (link 2) board.  I think is configured to take an input voltage up to 5V which it will then drop down to 3.3V.  If the SOT23 package the regulator mentioned then I would set the supply voltage to 5V as LDO regulators normally need a supply voltage greater than the output voltage.

I would then connect the BMP180 only to the Netduino `SCL` and `SDA` lines along with a common `GND` and then try to communicate with the board.

I would not attempt to connect the board to the Arduino UNO _unless_ you can verify that the `SDA` and `SCL` lines are working at 3.3V - if this is a true Arduino clone then I would expect it to be working at 5V.  The data sheet for the BMP180 states absolute maximum values for the supply to be 3.6V and the maximum input-high voltage to be the same as the supply voltage.

Regards,
Mark

vberkaltun, 2018-07-11:
Hi Mark,

Thanks for sparing time for me.

In this step, I believe that different voltage outputs is a very important subject for **`I2C Bus`**. In my opinion, Each device on a **`I2C Bus`** must have similar regulator circuit that can generate similar voltage output. Otherwise, the communacition can not execute. I have experienced this situation before on **`I2C`** Communication of NodeMCU with Arduino Pro Mini.

As you can see, I do not have good skills on circuit side, etc. but I try to learn it all with experiential learning and others.

More info:
[I2C Bus with 3.3v and 5.0v Device](http://forum.arduino.cc/index.php?topic=116345.0)
[I2C between 5V Uno and 3.3V Pro Mini](http://forum.arduino.cc/index.php?topic=165818.0)
[Pro Mini I2C not working](https://forum.arduino.cc/index.php?topic=358206.0)

Now, As you said, I'll try firstly BMP180 Sensor Module. if I can make it, I'll go next step.

Regards,
Berk

IOT-123, 2018-07-11:
Hi Berk,

I found something that may be of interest to you.

Using the original **i2c scanner** in **Netduino.Foundation.Core.Samples**, I could not find any of my ATTINY85s (on 3V3) using the debugger from Visual Studio. I went to use the **MFDeploy** tool this morning; it had the I2CScanner assemblies on it. When I connected the console output showed:
**Found I2C device at: 10** (the slave that was connected address). 

I do not have enough understanding of the toolchain as to why, but have options now...

Regards,
Nic