WildernessLabs/Netduino.Foundation #59: Piezo play tone won't play if already playing

bryancostanich, 2018-08-15:
I think we should re think this. If a tone is already playing, maybe we should just play the new tone? 

https://github.com/WildernessLabs/Netduino.Foundation/blob/3f24b67323daf915200f50cec614ff154d5e2355/Source/Netduino.Foundation/Speakers/PiezoSpeaker.cs#L26

adrianstevens, 2018-08-15:
The issue is the thread.Sleep call - we'd either have to abort the thread or sleep for sorter durations and constantly check if we need to stop the currently playing note. 

As it's written you can play tones two ways:
1. Call PlayTone with duration: is this case, it plays the tone for the time provided and you must wait for the time to elapse to play another note 

2. Call PlayTone without a duration: in this case, the tone plays until StopTone is called, so the coordinating code has complete control- if you need to interrupt tones, use this strategy

bryancostanich, 2018-08-15:
Good call.

closed 2018-08-15 by bryancostanich