WildernessLabs/Netduino.Foundation #65: [BME280+Adafruit ] Sample runtime crashes: "WriteRead: Retry count exceeded"

patridge, 2018-09-24:
Page: http://netduino.foundation/Library/Sensors/Atmospheric/BME280/

If I hook up my [Adafruit BME280](https://www.adafruit.com/product/2652) to a Netduino 3 Wi-Fi with the same I2C connection described (translating SCL to SCK and SDA to SDI on the [Adafruit version](https://learn.adafruit.com/adafruit-bme280-humidity-barometric-pressure-temperature-sensor-breakout/pinouts)), get the following exception when I run either the interrupt or polling examples on the page linked.

The exception comes from the constructor call: [interrupt](https://github.com/WildernessLabs/Netduino.Foundation/blob/087d7220aa3a58051ce59a8017e71f5e3539e283/Source/Peripheral_Libs/Sensors.Atmospheric.BME280/Samples/BME280InterruptSample/Program.cs#L27) or [polling](https://github.com/WildernessLabs/Netduino.Foundation/blob/087d7220aa3a58051ce59a8017e71f5e3539e283/Source/Peripheral_Libs/Sensors.Atmospheric.BME280/Samples/BME280PollingSample/Program.cs#L18).

> An unhandled exception of type 'System.Exception' occurred in Netduino.Foundation.dll
> 
> Additional information: WriteRead: Retry count exceeded.

I have not tried debugging from source yet, but that is my next step to get a source line inside the constructor. Additionally, I haven't tested my sensor outside Netduino, so I have no proof that the sensor isn't malfunctioning yet either. (I'll try to attempt that later, too.)

patridge, 2018-09-25:
I was able to test it against the NuGet source last night after resolving the `DebuggerBrowsableState` issue. It is failing on the [`ReadCompensationData` call to `ReadRegisters`](https://github.com/WildernessLabs/Netduino.Foundation/blob/087d7220aa3a58051ce59a8017e71f5e3539e283/Source/Peripheral_Libs/Sensors.Atmospheric.BME280/Driver/BME280.cs#L467). I'll try reading some datasheet info, but I'll probably end up trying to get it working against a non-Netduino device, too, to rule out a bad unit.

patridge, 2018-09-25:
Confirmed BME280 board is working against an Arduino board after fumbling through the [Adafruit guide](https://learn.adafruit.com/adafruit-bme280-humidity-barometric-pressure-temperature-sensor-breakout/arduino-test).

NevynUK, 2018-09-27:
> Confirmed BME280 board is working against an Arduino board after fumbling through the [Adafruit guide](https://learn.adafruit.com/adafruit-bme280-humidity-barometric-pressure-temperature-sensor-breakout/arduino-test).

Did you add pull-up resistors?

Regards,
Mark

patridge, 2018-09-27:
Oops...I definitely did not. Totally missed that note after the diagram. I'll go hunt down a resistor and give that a shot.

Thank you! I'll close this for now.

closed 2018-09-27 by patridge

patridge, 2018-09-27:
After a short time on Wikipedia, I got it wired up and tested. It worked great! Sorry for the missed instructions.
@NevynUK Is there something special about the Arduino boards that don't require the pull-up resistor?

NevynUK, 2018-09-27:
Firstly, glad to hear that you are up and running.

The subject of pull-up resistors is not trivial.  Some board have them built in, in this case, the SparkFun board does but the Adafruit board does not.  This is not the case for all Adafruit boards.  You have to check the schematic for the board you have purchased.

The 4K7 rule is good enough for single boards.  It is not a fixed rule, something around that value is good it is just that 4K7 is a common value that most people are likely to have.

For multiple boards then you can either calculate the value or use trial and error (a scope will help).

A couple of good sites:

[Effects of varying I2C pull ups](http://dsscircuits.com/articles/effects-of-varying-i2c-pull-up-resistors)

[I2C Design Mathematics](https://www.allaboutcircuits.com/technical-articles/i2c-design-mathematics-capacitance-and-resistance/)

Hope this helps,
Mark